splintered-reality/py_trees #32: Imposter's double initialise/terminate

naveedhd, 2017-01-04:
For behaviours wrapped in Imposter, the initialise and terminate are called twice.

```

[ INFO] Behaviour    :   Sync Maps [FileTransfer::initialise()]
[ INFO] Behaviour    :   Sync Maps [FileTransfer::initialise()]
[ INFO] Behaviour    :   Sync Maps [FileTransfer::update()]
[ INFO] Behaviour    :   Sync Maps [FileTransfer.terminate()][Status.FAILURE->Status.FAILURE]
[ INFO] Behaviour    :   Sync Maps [FileTransfer.terminate()][Status.FAILURE->Status.FAILURE]
```

naveedhd, 2017-01-04:
fyi @stonier

stonier, 2017-01-19:
Some analysis, for reference:

## PyTrees Axioms

Pathways used to call critical methods in a behaviour relevant to this discussion.

* `initialise()`
  * When a Behaviour’s or Composite’s tick starts and it is not currently running
[behaviour.py:228](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/behaviour.py#L229), [composites.py:254](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/composites.py#L254)
* `stop()`
  * When a Selector switches to a higher priority: [composites.py:271](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/composites.py#L271)
  * When a Behaviour’s tick results in not running: [behaviour.py:235](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/behaviour.py#L235)
  * When pruning a subtree, [trees.py:125](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/trees.py#L125), via Composite’s remove_child() in [composites.py:142](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/composites.py#L142)

## Current Imposter

### What

* Reflects non-tick pathways for initialise() and stop() directly to the child behaviour, [meta.py:109-11](https://github.com/stonier/py_trees_suite/blob/devel/py_trees/src/py_trees/meta.py#L109)
* Reuses Behaviour’s tick() which calls the child’s initialise() and stop() via reflected handles as normal

### Issues

* Imposter to Behaviour will call `initialise()` twice
* Imposter to Behaviour will change the Behaviour’s state based in Imposter’s state 
for eg. running_is_success -> running
* Imposter to Composites will call `stop()` twice

## Solution Proposal

Imposter implements its own tick which doesn’t manage `initialise()` and `stop()` for behaviour

With this solution, it won’t affect (skip) stop because the top-down `stop()` will still happen

### Rooftops

* **Pruning** : child will stop() via reflection ……….✔
* **Switch to Higher Priority**: child will stop() via reflection ……….✔
* **Potential Zombies**: imposter which returns Failure/Success, but child is still Running
  * Yes, can be, but will eventually get stopped at some point (1 & 2).....✔
* **Todo**:  Should be documented to avoid surprises for the unwary developer

…..can’t think of anything else….so green light!

commit 259de4bc (2017-01-20): Imposter's own tick  (#34)

* Bugfixing imposter's tick for better initialise/stop management, fixes #32 
* Some better logging here and there
closed 2017-01-20 by stonier