splintered-reality/py_trees #40: Selector Pattern - Protected Children

stonier, 2017-02-19:
This has been one I've been pondering for a while. When interrupted by a higher priority activity, it is quite natural to want to pick up the original where you left off when the higher priority activity is finished. Selectors dont let you do this though, you have to start again.

Here we just mark a child so that if it gets interrupted by a higher priority sibling while running (again, not from above), then it doesn't go to invalid. It keeps its running status which naturally resumes when the higher priority branch has finished.

naveedhd, 2017-03-06:
@stonier Consider the situation where Selector has a Sequence as it's first child(higher priority) and Sequence returns False. On the next tick of Selector, the Sequence will restart (because of of Invalid state) causing interrupt in any running behaviour after the Sequence. 

I think it's not intended by the Selector. Let me know if you need more details or want me to make a separate issue.

stonier, 2017-03-06:
Separate issue would be good. Answer the below there too. 

That interrupt should only happen if the sequence restarts (initialise())  and returns running or true in that first tick. If it is immediately returning false it should not interrupt. Is this happening?

stonier, 2019-01-17:
The original ask has been implemented by the `pick_up_where_you_left_off` idiom.

closed 2019-01-17 by stonier