splintered-reality/py_trees #46: Behaviour Index

stonier, 2017-02-23:
A page dedicated to just list up all the behaviours available in this package. 

* Should be in the 'reference' group
* Should just be a link and synopsis to documentation in the module api

naveedhd, 2017-06-27:
What is missing from the already listed in Module API section?

I am unable to see the synopsis in the sphinx docs but it's there when building docs locally. Is this an issue with the docs?

stonier, 2017-06-27:
Yes, indeed. Looks like autodoc has failed.

stonier, 2017-06-27:
module api now showing up after 4581b5d1 and 7bdbaaaf

stonier, 2017-06-27:
Was thinking of having a succinct list without all the other noise. It's far from urgent though and the module API probably really is enough.

closed 2017-06-27 by stonier