splintered-reality/py_trees #49: Sequence(Composite) under Selector

naveedhd, 2017-03-08:
*copied from other issue:* Consider the situation where Selector has a Sequence as it's first child(higher priority) and Sequence returns False. On the next tick of Selector, the Sequence will restart (because of of Invalid state) causing interrupt in any running behaviour after the Sequence.

*for the question:* That interrupt should only happen if the sequence restarts (initialise()) and returns running or true in that first tick. If it is immediately returning false it should not interrupt. Is this happening?
*answer:* Yes.

naveedhd, 2017-03-08:
If we have the following tree:

```
root = py_trees.Selector("Selector")

sequence_to_fail = py_trees.composites.Sequence("Sequence")
success = py_trees.behaviours.Success("Success")
failure = py_trees.behaviours.Failure("Failure")
running = py_trees.behaviours.Running("Running")

root.add_child(sequence_to_fail)
sequence_to_fail.add_child(success)
sequence_to_fail.add_child(failure)
root.add_child(running)
```

The ticking would be like:

```
--------- Run 0 ---------

[DEBUG] Selector     :   Selector [Selector.tick()]
[DEBUG] Selector     :   Selector [Selector.update()]
[DEBUG] Sequence     :   Sequence [Sequence.tick()]
[DEBUG] Sequence     :   Sequence [Sequence.update()]
[DEBUG] Behaviour    :   Success [Success.tick()]
[DEBUG] Behaviour    :   Success [Success::update()]
[DEBUG] Behaviour    :   Success [Success.stop()][Status.INVALID->Status.SUCCESS]
[DEBUG] Behaviour    :   Failure [Failure.tick()]
[DEBUG] Behaviour    :   Failure [Failure::update()]
[DEBUG] Behaviour    :   Failure [Failure.stop()][Status.INVALID->Status.FAILURE]
[DEBUG] Behaviour    :   Running [Running.tick()]
[DEBUG] Behaviour    :   Running [Running::update()]

--------- Run 1 ---------

[DEBUG] Selector     :   Selector [Selector.tick()]
[DEBUG] Selector     :   Selector [Selector.update()]
[DEBUG] Sequence     :   Sequence [Sequence.tick()]
[DEBUG] Behaviour    :   Success [Success.stop()][Status.SUCCESS->Status.INVALID]
[DEBUG] Behaviour    :   Failure [Failure.stop()][Status.FAILURE->Status.INVALID]
[DEBUG] Sequence     :   Sequence [Sequence.update()]
[DEBUG] Behaviour    :   Success [Success.tick()]
[DEBUG] Behaviour    :   Success [Success::update()]
[DEBUG] Behaviour    :   Success [Success.stop()][Status.INVALID->Status.SUCCESS]
[DEBUG] Behaviour    :   Failure [Failure.tick()]
[DEBUG] Behaviour    :   Failure [Failure::update()]
[DEBUG] Behaviour    :   Failure [Failure.stop()][Status.INVALID->Status.FAILURE]
[DEBUG] Behaviour    :   Running [Running.tick()]
[DEBUG] Behaviour    :   Running [Running::update()]
```

The above shows the `Success` and `Failure` under `Sequence` would run repeatedly.

For a more specific example of what we have, We have a docking controller under a Sequence and a Failure sound interaction (with running time of 1sec). When docking controller fails, it should return the sequence with failure and play the sound. But within the 1 second running time of 2nd child, the sequence is re-initialised and interrupts.

hope I this makes sense..

naveedhd, 2017-03-08:
I suppose this is due to explicit [Invaliding](https://github.com/stonier/py_trees/blob/devel/py_trees/composites.py#L479-L486) the child nodes under sequence.

stonier, 2017-03-08:
Your demo scenario is I think not exactly matching your problem. 

Firstly, re-initialising the sequence and its children (indirectly the children is a first step to paving the way for this) immediately is expected. Selectors will always pop into the highest level child every tick. However, this will only interrupt a running second child if the sequence returns running or success after a previous failure. If it's a failure after a previous failure (as in the simple example above), the second child won't get invalidated, it will just continue running.

In your docking controller example, it sounds like your second child is a sound behaviour of 1s duration. The only way this can be interrupted after a docking controller fail, is if it tries to start docking again and drops into a running state on the first tick after failure. This is exactly Selector behaviour.

What you probably want is to run the Chooser, which will initially select the docking controller until it fails, and then it will select the sound subtree on docking failure. It will then run the sound subtree through to completion. i.e. a Chooser hangs on to its selection until the selection ends in FAILURE|SUCCESS....or the chooser itself is interrupted from above.

naveedhd, 2017-03-08:
This is very clear and precise. Thank you very much.

closed 2017-03-08 by naveedhd

naveedhd, 2017-03-08:
Interestingly, the Chooser also return with FAILURE if the sequence child under is FAILURE:

[image]

stonier, 2017-03-08:
Ah this looks perhaps like a logic bug then. Will explore more later

stonier, 2017-03-24:
@naveedhd could you open an issue for this one? And if you'd like, take a shot at solving it?

naveedhd, 2017-03-28:
Yes I will do it.