splintered-reality/py_trees #54: Error when using meta.oneshot

jr-bor, 2017-05-02:
I will first explain my tree as a context, then the problem and my solution and then we can discuss if actually I am facing my tree correctly.

I want to have an "assisted move base". This means that the robot will try to go to a place and if it arrives we are happy. BUT if the has been any problem (i.e. goal blocked) then a message appears in a HMI where the user can press "cancel" or "retry". I could hardly come up with the idea represented in the image.
[image]
I have created a chooser to have the priority behavior. The first thing it does it to check a bb variable to see if it has to retry. If it fails, then it does the move base. Then two possible outcomes: (1) It succeeds or (2) it fails. If (2) happens, then it goes to the final state "ask_person", that basically ask the person and if this one answers with a "retry" then we write down that down into the bb variable. That will trigger the first behavior and thus "reset" the move_base. To me this makes kind of sense (although i dont have that much of that, so we can discuss this later). The only problem I have is that as I am using a selector, and then move_base gets "reset" all the time because it has more priority than "ask_person". I have solved this by wrapping the move_base with a meta.oneshot that keeps the last resulting state (FAILURE OR SUCCEEDED). Tree described.

In the moment I wrap the move_base with the meta.oneshot I get an error when ticking that says:
´´´
  File "/path_to_py_trees/py_trees/py_trees/composites.py", line 307, in tick
    for node in child.tick():
TypeError: wrapped() takes at least 1 argument (0 given)
´´´
By taking a look at the code I saw that the wrapper "replaces" the tick function (this explains the error in tick) with another function that returns the own node if we have got FAILURE or SUCCEEDED. The problem is that this needs _self_ as argument, and when it ticks this is not given. I have offerd the pull request #53 that fixes MY problem, but i feel that is not a good solution.

This said, I would like to know also what do you think about how to tackle this problem :smile:

jr-bor, 2017-05-03:
I found out my problem! It might help someone:

The problem was that I was doing
```
py_trees.meta.oneshot(this_is_a_behavior(with, parameters, in, here) 
```
instead of 
```
py_trees.meta.oneshot(this_is_a_behavior)(and, the, parameters, are, now, here!)
```
Subtle typo, but there it is.

closed 2017-05-03 by jr-bor

stonier, 2017-05-15:
Sorry for the delay, glad to see you sorted it out :)