splintered-reality/py_trees #62: parallel composite termination

joao-alex-cunha, 2017-07-07:
I'm having issues with getting a parallel node to terminate.
Consider the following situation. You have two behaviours A and B that run in parallel. At instant t behaviour A returns success while B is still running. At instant t+1 B returns success. This should make the parent terminate. However this isn't happening. Since the parallel composite ticks ALL its children, a children that terminates on a previous time instant is put running on the next time instant. The only way to make the parallel terminate is to have all children terminate on the same time instant. This should't be required.

A solution is to create a separate list that keeps track of running children. This list is populated with all the composite's children at the initialise method. Then only the running children are ticked for execution and removed from the list after they stop running.

stonier, 2017-07-07:
> This should make the parent terminate.

The kind of logic for that policy (`SUCCESS_ON_ALL`) is fine and fits certain use cases. What you are looking for is a different kind of logic. I think it's reasonable and likely to be a common use case. The most important point here is that I can't think of any easy way to implement what you'd like to do without creating an additional policy for parallels that caters for it.

If

stonier, 2017-07-07:
If you'd like to create a PR with a new policy (`SUCCESS_AFTER_ALL`?) and implementation for parallels, I'd be happy to review it.

naveedhd, 2017-07-14:
@joao-alex-cunha If the child A doesn't necessarily have to restart again after Success, you can decorate it in `meta.oneshot`. In this way it will keep it's status until child B finishes.

stonier, 2019-01-08:
Parallel discussion active and in #104.

closed 2019-01-08 by stonier