splintered-reality/py_trees #63: Design by YAML

naveedhd, 2017-07-14:
What do you think about the generating trees by parsing yaml file.

something like
```
root:
  name: Reality
  type: composites.Parallel
  children: child_a

child_a:
  name: Time
  type: timer.Timer
  duration: 5.0
```

I'm not sure if it would be useful for writing big but for small scripts (that I usually write to learn/verify) could be useful.

stonier, 2017-07-17:
I have no strong objections. Python should still be the first class citizen, but there are cases requiring serializable formats. YAML is ok, JSON might be an alternative so you can communicate with web apps (e.g. behaviour3's [JSON format](https://github.com/renatopp/behavior3js/wiki/JSON), check out the [editor](http://behavior3js.guineashots.com/editor/). 

Other thoughts:

* a parameter list for passing to the constructor
* include the package name, e.g. `py_trees.composites.Parallel`

stonier, 2018-10-12:
Just some notes: 

The lads here at TRI have been playing around with the idea of serialising the recipe for construction of a `py_tree` (note this is different than serialising the snapshot of a py_tree). This would serve some useful workflows:

* Export format from an editor
* Export format for converting from one form to another (e.g. we use for converting from log to scenario, where the scenario is represented by a behaviour tree)

Both of these workflows could, ostensibly, generate python code, but that is typically harder to do versioning on (and automatic upgrades of older version files lying around). Another benefit is that it could be used to generate the behaviour trees in a different language if ever this library decided to try it's hand at supporting it's implementations across languages (c++, javascript, ...).

facontidavide, 2019-05-29:
In BehaviorTree.CPP we did something similar, using XML as "scripting language".

To give you an idea of what it look like.... https://www.behaviortree.dev/xml_format/

erichlf, 2024-08-18:
I have created an xml parser for py_trees. It isn't ready to be released just yet, but my intention is to publish the code.

erichlf, 2025-01-31:
It is now published at github.com/sam-xl/py_trees_parser