splintered-reality/py_trees #65: Running in Python 3

aadeshnpn, 2018-01-12:
Py_trees doesn't work in Python 3 but I didn't see anywhere in the docs that it only runs in python 2.7.  What is true?

aadeshnpn, 2018-01-12:
Below is the error message
[image]

stonier, 2018-01-16:
Absolutely right. Hitherto we have not run, nor tested on Python 3 (nor had the need to).

Thanks for #66. I'll take it for a run with all the tests and demos on the weekend.

aadeshnpn, 2018-01-16:
I have already made the changes for python3 and submitted a pull request. You can verify the pull request.

stonier, 2018-01-22:
Thanks, now merged...but only on the devel branch right now.

closed 2018-01-22 by stonier

OlgaPaw, 2018-05-08:
When new version is planned to be released?

stonier, 2018-05-10:
@OlgaPaw are you looking for the python 3 functionality specifically? And release in pip or ROS?

aadeshnpn, 2018-05-14:
@OlgaPaw you can use my fork version of py_trees. It works with Python3

stonier, 2018-05-15:
0.6.0 is now up on pypi, with python2/python3 support. We had to take some additional care of the enum34 dependency.

Looks like we'll be using it at my new company too, so I'll be tinkering on this more frequently!