splintered-reality/py_trees #67: Implementing Repeater decorator

aadeshnpn, 2018-01-26:
I am trying to implement Repeator decorator using py_trees.

`def repeter(cls):
    def _update(func):
        @functools.wraps(func)
        def wrapped(self):
            while True:
                self.original.status = cls.update(self)
                if self.original.status == common.Status.FAILURE:
                    break
            return common.Status.SUCCESS
        return wrapped

    Repeter = create_imposter(cls)
    setattr(Repeter, "update", _update(Repeter.update))
    return Repeter` 

Please help to to implement in a better way.

stonier, 2018-02-08:
The original class' update will already be called by the imposter's tick, so no need for `cls.update(self)`. That is probably the primary source of your problems.

Be careful with the idea of a repeater. What you have here is a behaviour that can get 'stuck' and stop the entire tree for ticking over. What you are looking for instead, is a decorator that returns `RUNNING` unless the original returns `FAILURE` at which point the repeater will return `FAILURE` or `SUCCESS` depending on what logic you want. If you introduce a max no. of repeats, then `FAILURE` and `SUCCESS` of the repeater takes on real meaning.

aadeshnpn, 2018-02-09:
Thanks. After going few papers and resources, the behavior tree has lots of decorators dealing with RepeatUntilTrue, RepeatUntilFalse and others. So I wanted to have those decorators in this implementation as well.

stonier, 2018-02-10:
Indeed, they came up on the whiteboard when we were designing tree logic often. Just never ended up using them. Send us a PR if you get an implementation working!

closed 2018-02-10 by stonier