splintered-reality/py_trees #68: Is a graphical editor planned?

pirobot, 2018-01-31:
Great to find this awesome implementation of behavior trees.  Are you planning to develop a graphical editor for building trees?  If not, I was thinking of try to find someone to make one.  There are some really nice BT editors for the Unity game engine and they make it easy to construct complex behaviors, especially for robotics.

stonier, 2018-02-08:
Always thought about it, but never became an itch that we needed to scratch. We always had developers and they were more than happy working with the python and the offline tools (e.g. dot graph generator).

It'd certainly be a plausible thing now that this library has stabilised. A good first step might be to have a json serialisation for the trees. That shouldn't be a stretch given that the ros serialisation has I think, most of the essential information. This could then be used for a web app editor as well as a web version of rqt_py_trees.

If it's json, might be worth forking and bootstrapping from what the behaviour3editor has. Their feature set is quite smaller, but the editor looks nice.

pirobot, 2018-02-09:
OK thanks.  I may look into getting someone to do this.  I agree that working in python and using the dot graph generator is sufficient for most cases.  However, I have played with some of the BT editors in Unity3D when trying to build some complex trees and the drag and drop is rather nice. :)

closed 2018-02-09 by pirobot

M-Rahul1, 2024-03-22:
Did you manage to find BTeditor using pytrees?