splintered-reality/py_trees #69: Basic question about status colors in rqt_py_trees

pirobot, 2018-02-05:
I am running the tutorial_two.launch example and when the battery is OK, the "Battery Check" selector is **red** in rqt_py_trees because of the py_trees.meta.success_is_failure decorator applied to it.  I understand why the decorator is needed for the parent Priorities selector to work properly, but I find the red color of the "Battery Check" in rqt_py_trees counter-intuitive when in fact the battery is OK.

What I would prefer to see is: if the battery is OK, then "Battery Ok" is green and the "Battery Check" is green.  When the battery goes below threshold, I would like to see "Battery Ok" turn red and "Battery Check" turn red.  What combination of selectors, sequences, parallels, and metas would I need to get this behaviour?

stonier, 2018-02-08:
You can (I think) do it without decorators by converting the check to a parallel and switching the idle/flash behaviours:

```
...
battery_check=py_trees.composites.Parallel("Battery Check")
...
priorities.add_children([battery_check, flash_led_strip])
battery_check.add_children([is_battery_ok, idle])
```

The logic is correct in both cases so I would claim that the gui should find a way to intuitively represent both cases. You're right, since the gui doesn't support showing decorators very well at the moment, it can be a little confusing. That was always a todo, but we got used to shaping the debug text like a question and the green/red state like a yes/no answer. This worked even for test engineers.

I would like to have a decorated selector still look like a selector (octagon) but with additional circumventing rings for each decorator, along with a tooltip for each. Then you could show green on the inside and red on the outside for instance.

pirobot, 2018-02-09:
Thanks for the example--that helps a lot, and the status colors now look more intuitive to me at least.

closed 2018-02-09 by pirobot