splintered-reality/py_trees #83: Does not work on Windows?

fredrikbaberg, 2018-07-27:
Hi,
I'm trying to run some code on Windows 10, which works on Ubuntu.

When trying to run, I'm getting errors that modules are missing. First it was `fcntl`, which I "solved" through a dummy file (as in https://stackoverflow.com/a/25471508). Now it's `termios`, which I can't find a solution for.

Is it possible to modify py_trees to also run on Windows?

OS is Windows 10, Python 3.6 installed using Anaconda (x86_64).

stonier, 2018-08-03:
Never tried it on windows, but it should not be much work to get it there. I don't have, nor use `py_trees` on windows, so ideally it would need another developer to step up and make sure it is always tested and works beyond the unit tests and hello world examples.

Having said that, feel free to float a tentative PR that we could use to discuss where the code would need to be modified. 

In this case, these imports are only used in `read_single_keypress` which is in turn only used in the tutorials. Probably the cleanest solution (rather than trying to exactly substitute for `fcntl` and `termios`) would be to just have two separate implementations for `read_single_keypress`. Many instances of that abound on a google search.

fredrikbaberg, 2018-10-02:
Sorry for the (really) delayed reply, I used my Ubuntu install instead, to save some time.
I guess that means something like http://code.activestate.com/recipes/134892/ ?

Not sure if/when I'll have time to look at it, but if I manage to get it to work I'll create a PR.

stonier, 2018-10-12:
Status:

* Keypress functionality is in (caveat: key presses need a carriage return)
* No colouring

This gets us to a working functionality, albeit lacking some polish, so closing this issue. Feel free to create issues for the polish and put them on the ToDo milestone if you wish.

closed 2018-10-12 by stonier