splintered-reality/py_trees #86: Execution Bug using a oneshot decorator for a Sequence that affects a sibling Selector

sachit-tri, 2018-08-14:
Please see the attached examples provided here.

The differences between these files is on line 19 where a sequence has a oneshot decorator on it.

If you run the test py_trees_fail.py you will notice that the selector node going by the name "Selector" does not tick the Failure node going by the name  "Failure" after the first tick, even though it is supposed to.

Removing the oneshot decorator on the prior sequence node as shown in the test py_trees_success.py makes this work correctly.

[py_trees_bug.zip](https://github.com/stonier/py_trees/files/2287807/py_trees_bug.zip)

stonier, 2018-08-15:
The minimal example is (switch out the commented line for fail/success):

```
#!/usr/bin/env python

import py_trees
import time

root = py_trees.composites.Parallel(name="Parallel")

running = py_trees.behaviours.Running(name="Running")
#sequence_1 = py_trees.meta.oneshot(py_trees.composites.Sequence)(name="Sequence 1", children=[running])
sequence_1 = py_trees.composites.Sequence(name="Sequence 1", children=[running])

failure = py_trees.behaviours.Failure(name="Failure")
sequence_2 = py_trees.composites.Sequence(name="Sequence 2", children=[failure])

root.add_children([sequence_1, sequence_2])

bt = py_trees.trees.BehaviourTree(root)
bt.setup(0)

py_trees.logging.level = py_trees.logging.Level.DEBUG

for i in range(1, 3):
    try:
        print("\n--------- Tick {0} ---------\n".format(i))
        bt.tick()
        print("\n")
        py_trees.display.print_ascii_tree(bt.root, show_status=True)
        time.sleep(1.0)
    except KeyboardInterrupt:
        break
```

stonier, 2018-08-15:
**Root Cause**: Once the meta-code modifies a class, it *stays* modified.

stonier, 2018-08-20:
So, the decorator approach is, fundamentally correct and does fit the use case where you want to create a class and then apply a decorator to avoid writing additional boilerplate.

As noted though, it does not fit the dynamic use case where you want to modify the instance.

Users though, need eyes wide open when it comes to decorators. At the very least, documentation and FAQ should be updated.

@sachit-tri Until I get something in which permits the dynamic instance use case, drop the following code in the application side:

```
@oneshot
class OneShotSequence(py_trees.composites.Sequence):
   pass
```

stonier, 2018-08-20:
Closed by the above PR. Forthcoming in release 0.6.1.

closed 2018-08-20 by stonier