splintered-reality/py_trees #133: Exception Handling in 1.0

stonier, 2019-01-14:
* `AssertionError` only for errors that should never happen. Note that assertion errors can be optimised out, so this does not include checking user inputs.
* Type/RuntimeErrors only or create some customer exceptions?
* Should setup return an exception instead of true/false?

FYI: @sachit-tri @naveedhd

stonier, 2019-01-14:
Q: Can type hints transfer to docstrings?

sachit-tri, 2019-01-16:
> AssertionError only for errors that should never happen. Note that assertion errors can be optimised out, so this does not include checking user inputs.
:+1: 

> Type/RuntimeErrors only or create some custom exceptions?
I think we should stick to these for now/

> Should setup raise an exception instead of returning true/false?
Honestly, I don't think there's a need to return true / false. In the case of an error during setup a node must throw the appropriate exception -> TimeoutError or whatever else.

>  Type hints are nice, but do not raise errors in runtime
I don't think this should be too much a problem.

closed 2019-01-17 by stonier