splintered-reality/py_trees #151: PyTree Creator / Monitor

stonier, 2019-01-21:
Not exactly the right place for this (this is not the gui repository), but just to collect resources. Some nice guis:

* [BehaviourTree/Groot](https://github.com/BehaviorTree/Groot) - c++ creator, maybe monitor as well?
* [Behaviour3Editor](https://github.com/behavior3/behavior3editor) - web based creator?

stonier, 2019-01-21:
Note: right now we have rqt_py_trees which is a ROS runtime monitor. 

A creator could/should be agnostic to ecosystem, the monitor really could be as well, so long as the backend for it is pluggable (or perhaps uses its own communication mechanism).

naveedhd, 2019-02-20:
somewhile (almost an year) ago, mused upon this idea doing with [rqt](https://github.com/naveedhd/rqt_py_trees_creator). couldn't figure out at that time the following:

- in my head was drag and drop thing with available behaviours, couldn't figure out straightaway example of doing this
- if it has to be (to more extent) agnostic of py_trees/ros (lets just say we pass around json or some other message type for tree definition/construction) then there could be other tools (web/platform). The only thing that it still needs to lookup from package(s)/directories for existing behaviours and their parameters.

sachit-tri, 2019-02-20:
The project I really seem to like (at first glance) is https://github.com/pruttned/owl-bt. My rationale is that the configuration is entirely json based so it makes it very easy to author trees and behaviours. This json file should also be easy to auto generate.

Regarding monitoring with this task, I don't know. I haven't dived deep enough to understand how hard that would be.

facontidavide, 2019-05-29:
Hi

I am looking into your library to understand the relationship with BehaviorTree.CPP and the differences.

About using Groot... I like to say that Groot is a fancy XML editor.
If you (we?) implement a factory to create programmatically a tree from [this file format](https://www.behaviortree.dev/xml_format/), py_trees might become compatible with Groot

stonier, 2019-06-02:
@facontidavide I'm getting pretty close to the point where I'll start tickling this issue - the ros2 port had more speed bumps than expected, but the light at the end of the tunnel is there. I'll reach out when I'm done.

I expect not too many changes to make it compatible with that schema, at least for creation It might need some tickling of the schema as well. 

I am curious though - do you use the same schema for both creation and monitoring?

The variables used for creation of the trees would be irrelevant for the monitoring task, and likewise the monitoring task carries much data about the behaviour that is independent of creation (status, feedback messages, etc).

facontidavide, 2019-06-02:
For monitoring I use serialization using flatbuffers and ZeroMQ for the transport mechanism.
This is a PITA, because it requires more work, but it is more efficient.

stonier, 2019-07-07:
Some notes...

* Groot makes use of a great library of renderable qt graph nodes, in c++. 
* Hard to move away from PyQt though, it's just so much easier

On the web app side, there are lots of pain points:

* The hassle of doing interactive widgets (e.g. buttons) in javascript
* The need for additional frameworks
* The need for a server
* Filesystem access complications
* Maintenance of libraries that seem to change yearly
* Additional bridges to ROS

Interesting options for web apps:

* [Electron](https://electronjs.org/) is a nice desktop wrapper around js apps (uses chromium and node internally). Lots to learn though and the web app has to handle everything. You'd also need an extra bridge to communicate out to the web app. Some experience at TRI with these though.
* QWebEngine with PyQt is also a desktop friendly option for js apps as well. It now uses chromium's webkit, not the original qt webkit which always struggled to keep up. You could create a rendering javascript library to render trees and execute that inside a QWebEngine widget. This ends up being rather useful - other people can use the rendering library to create their own custom web apps - it's unlikely the app interface I create here is a one-shoe-size-fits-all, but the rendering library certainly is. Another benefit, it would integrate easily inside ROS workspaces as a developer tool and spin off into a web app rather easily (as described above) for a productised or cloud served version.

stonier, 2019-07-07:
**The Javascript World**

* [createjs](https://createjs.com/) - used by the behaviour3 editor, canvas editing plus other goodies to boot
* [pixijs](https://www.pixijs.com/) - 2d webgl rendering
* [viz.js](https://github.com/mdaines/viz.js) - dot graphs
* [dagrejs](https://github.com/dagrejs) - directed graph layouts

stonier, 2019-07-21:
Closing this out, work started over in [py_trees_ros_viewer](https://github.com/splintered-reality/py_trees_ros_viewer/blob/devel/README.md).

closed 2019-07-21 by stonier