splintered-reality/py_trees #152: Tree change checks

stonier, 2019-01-21:
These are currently in py_trees_ros, but the logic should be here so that people can plug there own callbacks to handle 'a change in a tree' (e.g. logging mechanism). This library could even provide a default mechanism for logging.

stonier, 2019-01-21:
FYI @sachit-tri

stonier, 2019-02-02:
Musings. 

A callback _and_ a helper that serialises the tree (so the user doesn't have to write their own in most cases), but let the user write their own also if they do wish. 

The tree, too, is not going to be sufficient for explainability. You'll need a serialised dump of the blackboard at that point as well.

stonier, 2019-02-06:
**What Defines a Change in the Tree?**

The underlying motivation is to catch `moments` critical for understanding / diagnosing the decision making process. It should not a vehicle for overwhelming the user with tick-by-tick data changes. For that, implement post-tick handlers that take snapshots of the data, i.e. the blackboard.

To illustrate, given a battery monitoring behaviour, it pertains to a crossing of a threshold leading to an action (e.g. change in LED colour), not about the updated % at each tick.

**Implementation**

The first and obvious mechanism is to:

* Flag a `change` whenever the `status` of any behaviour in the tree changes
* Save `status`, `feedback_message` and `visited` of every behaviour as a snapshot of the tree state

Here, the `feedback_message` helps immensely with diagnosing problems, just as exceptions have string messages in most every language. Also `visited` is a boolean that identifies whether that behaviour was visited in the tree tick or not.

This approach comes up short when diagnosing problems in practice, because not *all* decision making is embedded in the tree. Often a behaviour is responsible for triggering and monitoring a long running action on a robot, e.g. docking and there is no way to decompose that more finely on the tree without breaking up the actual c++ docking control implementation, something that might be outside of your control. The docking controller itself will be moving through some internal state engine of it's own (and subsequently making decisions) and you'd like to reflect this back into the changes flagged by the tree. This is especially useful for monitoring programs when you've started using the tree monitor as a way of concisely determining what it's doing at any point in time. 

The pragmatic approach we used for a couple of years:

* Flag a `change` whenever the `status` of any behaviour in the tree changes
* Flag a `change` whenever the `feedback_message` of any behaviour in the tree changes
* Save `status`, `feedback_message` and `visited` of every behaviour as a snapshot of the tree state

The caveat here though is, if someone sets a `feedback_message` with, e.g. the battery level %, then it will spam the log, or the monitoring program with change information that makes it next to impossible to find those `moments` important to decision making. Whilst in a small group, this is easily enforced. However, I'm inclined to believe that the effort in enforcing this convention and reacting on reviews and in runtime to the inevitable spam bugs, ultimately makes this impractical.

An alternative would be to add an additional boolean flag to the `Behaviour` class that explicitly marks changes to be caught (when there has been no status change) and leave the feedback_message to be freely used in whatever way the user wishes. I typically endeavour to refrain from adding functionality to `Behaviour`, but it's probably warranted here.

@naveedhd, @sachit-tri thoughts?

sachit-tri, 2019-02-06:
When I think of change in a tree, the only thing I think of is the change in the state of one or more behaviours in the tree.

Having used feedback message as a debug tool I can understand why I would like to snapshot the tree and save it out for debugging purposes.

I am not sure what you mean by 'visited' here since I don't see any parameter on the behaviour regarding 'visited'. I think that should be implicit given that it would usually entail some change in either the state or the feedback message of a behaviour.

Here is my proposal:
By default we should only consider 'change' as a change of state of one or more behaviours in the tree. We should additionally support 'debug-change' or a better named type of change that handles feedback message changes as well.

stonier, 2019-02-06:
I do need something for the `long-running control component with internal decision making` use case so the current capability residing in `py_trees_ros` doesn't regress. For that, the `flag_changed` boolean will suffice and not get in the way for those who do not care to use it. Note: `debug-change` would help solve our use case (give us freedom to do with feedback_messages as we like) but not improve the framework in a way that eliminates the need for a human convention in others' use cases. We may also end up deciding we would like that too.

As to your question, `visited` derives from the information picked up by a (snapshot) visitor on the tree when you tick. It is not stored in the behaviour. Knowing the tick trace through the tree is very useful for debugging and without this tidbit, you'd have to infer this yourself from statuses which gets tedious on large trees for lazy software programmers.

sachit-tri, 2019-02-07:
Okay -> the flag_changed boolean sounds good to me. Maybe debug-change can just be based on feedback_message and flag_changed?

naveedhd, 2019-02-11:
I agree with the `side-effect` of feedback message (e.g. in case of battery level or some constant switching)  that it may spam the logs/viz. 

However `flag_changed` have the tendency of being ignored in most cases until required. In my observation, I have seen even feedback_message not utilized properly. So that increases the things to keep in mind for developer for every behaviour that she writes.

[Musing](https://github.com/stonier/py_trees/issues/152#issuecomment-459984221) sounds appealing to me! would like to give this a thought.

stonier, 2019-02-12:
Had a rethink after looking at my cheap and dirty implementation in `py_trees_ros` which couples the idea of change with the ROS serialisation of the tree. I'm also trying to get around having to add something awkward to the base class if I don't need to. Some ideas:

1. Use a visitor (that only visits the ticked part of the tree) to look for change

It shouldn't need to visit the entire tree, so this scales better. All it has to do is save a dict of the tree id/status and compare it with a saved dict from the last visitation. In `visitor.initialise()`, The visitor would have a flag for `changed` set to False and in `visitor.run()` if any behaviour causes the new dict to differ, it toggles the flag.

2. Ideally someone who wants to log needs two things: a) the tree to mine for logging information and b) the visitor details to know what part of the tree got visited

3. Someone who wants to detect change can write their own custom visitor to do the job. Ours need only be the 'common' example.

4. `BehaviourTree` perhaps doesn't need any api changes itself. The owner of the tree need only pass in a `post_tick_handler` for logging, but in that case, we'll need to store visitors in a dict, not a list, so the `post_tick_handler` can find the change visitor.

Actions:

* [ ] a change visitor (need a cool name!) that acts on status changes
* [ ] update `BehaviourTree` to store visitors in a dict with a keyword identifier, not a list
* [ ] tests
* [ ] demo and docs

PR #159 (2019-02-13): [visitors] winds_of_change visitor, with logging demo
Resolves #152.

* [x] a change visitor (need a cool name!) that acts on status changes
* [ ] ~~update `BehaviourTree` to store visitors in a dict with a keyword identifier, not a list~~
* [x] tests
* [x] demo and docs

The second wasn't necessary since `functools.partial` is sufficient. Subsequently, this can be a patch version improvement.
closed 2019-02-13 by stonier