splintered-reality/py_trees #162: decorator.py not in release 0.5.11

simchanu29, 2019-02-25:
In the branch `release/0.5.x` the file `py_trees/decorator.py` exists
In the release [0.5.11](https://github.com/splintered-reality/py_trees/archive/0.5.11.zip) this file doesn't exists

It leads to the following error when using the first tutorial of the `py_tree_ros` ROS package : 
```
  Traceback (most recent call last):
  [...]
  [...] py_trees_ros/py_trees_ros/conversions.py", line 49, in behaviour_type_to_msg_constant
    elif isinstance(behaviour, py_trees.decorators.Decorator):
AttributeError: 'module' object has no attribute 'decorators'
```

simchanu29, 2019-02-25:
I've just tested the `py_trees_ros` package with the branch `release/0.5.x`. I get the error :
```
Traceback (most recent call last):
[...]
[...] py_trees/py_trees/display.py", line 303, in stringify_dot_tree
    graph = generate_pydot_graph(root, visibility_level=common.VisibilityLevel.DETAIL)
TypeError: generate_pydot_graph() takes exactly 3 arguments (2 given)
```
I've checked the code, it seems there is indeed an error on the number of arguments line 303 of `display.py`

After a checkout to version `0.6.8` it seems to solve the issue. 

So either the advised version for ROS Kinetic should to be changed to `0.6.x`, or a fix should be made in `display.py`.

stonier, 2019-02-25:
Thanks for the catch...getting that experimental feature in across kinetic/melodic and crystal was messy. I've been trying to consolidate the branches.

Cherry-picked https://github.com/splintered-reality/py_trees/commit/f3a37ae96feb9fcd758e023f822d2ba030e406ed into release/0.5.x and releasing 0.5.12.

stonier, 2019-02-26:
Kinetic release [v0.5.12](https://github.com/ros/rosdistro/pull/20378) on it's way.

closed 2019-02-26 by stonier

naveedhd, 2019-02-27:
@stonier I think 0.5.2 broke with warnings and still not released (so the problem persists). this [pr](https://github.com/splintered-reality/py_trees/pull/163) might solve the break.