splintered-reality/py_trees #186: Is py_trees backwards compatible with older ROS?

jstyrud, 2019-03-20:
Quick question that I couldn't tell from the readme. Is the main py_trees code backwards compatible with older ROS versions?
For example, ROS1/Kinetic is listed as compatible with py_trees 0.5.x. Does that mean that Kinetic works **only** with 0.5.x or that it works with **0.5.x and higher**? (or simply that it's only tested with 0.5.x and it might work with later versions but we don't know?) 

The question arises because I need to use Kinetic but I also want to use the latest version of py_trees because of some new stuff that's in there.

stonier, 2019-03-22:
Once I release on a distro, I typically do not bump the minor version, just to keep stability for users of the package. That means the deb will stay on the `0.5.x` release. However, the dependencies are so minimal that source releases for other versions typically work. In this case, the melodic `0.6.x` release will (should) work fine on kinetic if you download the sources and put that in your workspace.

Having said all that, beyond `0.6.x` there was a shift to drop python2 support so I could take advantage of the python3 without being constrained by having to keep it python2 compatible. I was not using the library in a python2 environment in the company any longer either, and that made it harder to be sure the compatibility was sound (tests are great, but they always miss a few things if you don't have a complex use case behind it).

Which features were you looking at? Perhaps we can do some backporting. @naveedhd recently helped bring the revamped decorators back to `0.5.x` and `0.6.x`.

stonier, 2019-03-22:
Note to self: the Readme needs some grooming and could use a note about python2/python3 support.

stonier, 2019-04-29:
@naveedhd is doing some work to backport newer features across to a kinetic-compatible (read python2 compatible) branch so that a jump to the 1.x versions later is minimal.

stonier, 2019-05-13:
Aging out.

closed 2019-05-13 by stonier