splintered-reality/py_trees #200: Default synchronise on ParallelPolicy.SuccessOnAll should probably be False

andrewbest-tri, 2019-06-21:
It is confusing that children of a parallel are not restarted by default if any other child returns true. I would expect them to be ticked always unless I specifically tell them not to.

stonier, 2019-06-24:
I wonder if this is one of those things where confusion is in the eye of the beholder - more specifically, in light of what they are trying to achieve. It's only recently py_trees has had the ability to synchronise, and it's failure not to do so was surprising to a few (I've had reports anecdotally, also see #62).

Perhaps I should rejig it so that it has no default? That will make sure devs don't end up in a debugging spiral when incorrectly assuming the default behaviour.

andrewbest-tri, 2019-06-24:
I can live with that. It seems like wherever one's background in behavior trees comes from, they tend to expect different things.

closed 2019-10-17 by stonier