splintered-reality/py_trees #230: Built-in test tools

arthurrichards77, 2019-09-20:
We're using py_trees for drone control, and we'd like to be able to run the tree in "test mode" to verify its logic without firing up the drone interface or bothering with a simulator.  I want the test capabilities built into the tree, rather than having a second tree builder with test leaves that risks getting out of step - test what you fly, etc.
 
I'd therefore like to have a "test injector" decorator that, when enabled, ignores its child and returns either a random status or a predefined test status.  I'd also like "assert" decorators to verify requirements on components, e.g. AssertNeverFails(), and "coverage" tools (either as a Visitor or as a Decorator) that just monitor if a test run has adequately stimulated the leaf nodes (via their test injectors).

I'm some of the way through implementing these in a fork.  What do you think?

stonier, 2019-09-30:
I'd be interested in seeing what you have - can you throw me a link? I believe I've read of something similar in a graphing / tree framework - were you inspired from something else?

I've always done such testing by creating a mock robot under the tree and creating integration tests with that (or subtrees thereof) to make sure I cover all of the permutations and combinations that needed regression tests. I partly based the ros tutorial documentation around this concept - https://py-trees-ros-tutorials.readthedocs.io/en/release-1.0.x/. Of course, not as complex as what we had in-house, but the flavour is there.

stonier, 2019-11-12:
Aging out.

closed 2019-11-12 by stonier