splintered-reality/py_trees #270: `children=[]` in constructor (release/0.5.x)

nathantsoi, 2020-01-19:
i'm using `release/0.5.x` w/ ros and have found the constructor version of `children=` does not work. adding children via `add_children` works, but using this is a bit strange as it requires building the tree (in code) from the bottom up. any suggestions what's going on w/ the order of calls when using the `children` parameter in the `composites` constructor?

stonier, 2020-01-20:
Not aware that anything with it is broken. Certainly the code in `release/0.5.x` looks fine.

https://github.com/splintered-reality/py_trees/blob/42d73682b5118a6edba2570563b188c1d67ab4d2/py_trees/composites.py#L64-L67

The `devel` branch dropped the map but should be functionally the same.

https://github.com/splintered-reality/py_trees/blob/c4eceb5005684c493255b514eb1b8cc37dbfee7e/py_trees/composites.py#L64-L68

Do you have a code snippet for which it goes wrong along with the error message?

nathantsoi, 2020-01-20:
thanks @stonier, i agree, i saw that code, it looks fine. i also [added some tests](https://github.com/nathantsoi/py_trees/commit/102ae8ff04aad1131ab43b1d84e91b33f2ab6268), all pass: 

however, with this change:
``` def create_root():
-    root = py_trees.composites.Parallel("Tarzan")
     standby_or_interact = py_trees.composites.Selector("Priorities")
     standby_or_interact.add_children([
         create_failsafe_check(),
         create_interaction(),
         create_standby()
     ])
-    root.add_children([
+    root = py_trees.composites.Parallel("Tarzan", children = [
         create_topics2bb(),
         standby_or_interact
     ])
```

I go from this (correct) debug output:

```
[DEBUG] Tarzan               : Parallel.setup()
[DEBUG] TopicsToBB           : Sequence.setup()
[DEBUG] Priorities           : Selector.setup()
...
```

To this (incorrect debug output):

```
[DEBUG] Tarzan               : Parallel.setup()
[DEBUG] Tarzan               : Parallel.tick()
[DEBUG] Tarzan               : Parallel.stop()[Status.INVALID->Status.SUCCESS]
Traceback (most recent call last):
  File "/project/scripts/tree_node.py", line 184, in <module> main()
  File "/project/scripts/tree_node.py", line 179, in main behaviour_tree.tick_tock(500)
  File "/py_trees/trees.py", line 259, in tick_tock self.tick(pre_tick_handler, post_tick_handler)
  File "/py_trees/trees.py", line 233, in tick node.visit(visitor)
  File "/py_trees/behaviour.py", line 217, in visit visitor.run(self)
  File "/py_trees_ros/visitors.py", line 108, in run self.tree.behaviours.append(conversions.behaviour_to_msg(behaviour))
  File "/py_trees_ros/conversions.py", line 116, in behaviour_to_msg tip = behaviour.tip()
  File "/py_trees/composites.py", line 124, in tip
    return self.current_child.tip() if self.current_child is not None else None
  File "/py_trees/composites.py", line 618, in current_child
    return self.children[-1]
IndexError: list index out of range
```

As you can see, I'm also using `py_trees_ros` (release/0.5.x) with ROS1 (melodic). Also, I'm using python 3. Any idea what I'm doing wrong here?

nathantsoi, 2020-01-20:
update: i'm not sure why, but https://github.com/nathantsoi/py_trees/commit/6ae9e0ac00fc8393b762585363ca250c57f38b8d appears to fix the problem. any idea how i can cover this with a test?

stonier, 2020-01-23:
Aha, just noticed you mentioned you were using python3. The use of `map` here got deprecated for python3 (668baa92). 

Reason being that maps now produce a generator. If you don't actually use the result of a map call (we don't even assign it to anything), it will never generate anything and subsequently never executes the function you're mapping to. Hence your tree above collapses to just the root behaviour.

What you can do - any reason you are using `v0.5` on melodic? Some python3 compatibility went into `v0.6` [release/0.6.x](https://github.com/splintered-reality/py_trees/blob/release/0.6.x/py_trees/composites.py#L62), though I quickly pulled out of officially supporting both since I'd stopped using python2/melodic.

stonier, 2020-02-03:
> any idea how i can cover this with a test?

Just call that constructor with a few children and check that the length of the `Behaviour.children` variable is equal to the number you added. In the former, with `map` and `python3`, you'll get zero. In the latter, you'll get the correct number of children.

Then shoot a PR to the appropriate release branch.

stonier, 2021-01-10:
Aging this one out (not planning to work on p3 compatibility for old versions). Accepting PR's for it though.

closed 2021-01-10 by stonier