splintered-reality/py_trees #287: IndexError on empty tree with unicode_tree

andrewbest-tri, 2020-05-08:
Repro:
```
tree_root = py_trees.composites.Parallel(
    name="ScenarioTree", policy=py_trees.common.ParallelPolicy.SuccessOnAll(synchronise=False),
)
tree = py_trees.trees.BehaviourTree(root=tree_root)
tree.setup()
tree.tick()
```

expected:
```
'[||] ScenarioTree\n'
```

actual:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-22-547e4ecd5b7b> in <module>
----> 1 py_trees.display.unicode_tree(tree.root)

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/display.py in unicode_tree(root, show_status, visited, previously_visited, indent)
    284         previously_visited,
    285         indent,
--> 286         symbols=ascii_symbols
    287     )
    288     return lines

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/display.py in _generate_text_tree(root, show_status, visited, previously_visited, indent, symbols)
    127     if symbols is None:
    128         symbols = unicode_symbols if console.has_unicode() else ascii_symbols
--> 129     tip_id = root.tip().id if root.tip() else None
    130 
    131     def get_behaviour_type(b):

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/composites.py in tip(self)
    102             :class::`~py_trees.behaviour.Behaviour`: the tip function of the current child of this composite or None
    103         """
--> 104         if self.current_child is not None:
    105             return self.current_child.tip()
    106         else:

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/composites.py in current_child(self)
    669                 if child.status == common.Status.SUCCESS:
    670                     success_children.append(child)
--> 671             return success_children[-1]
    672         else:  # RUNNING
    673             return self.children[-1]

IndexError: list index out of range
```

andrewbest-tri, 2020-05-08:
May be deeper
```
tree_root = py_trees.composites.Parallel(
    name="ScenarioTree", policy=py_trees.common.ParallelPolicy.SuccessOnAll(synchronise=False),
)
tree = py_trees.trees.BehaviourTree(root=tree_root)

tree.setup()
tree.tick()
new_thing = py_trees.behaviours.Success("foo")
new_thing.setup()
tree.root.add_child(new_thing)
py_trees.display.unicode_tree(tree.root)
```
output:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-31-547e4ecd5b7b> in <module>
----> 1 py_trees.display.unicode_tree(tree.root)

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/display.py in unicode_tree(root, show_status, visited, previously_visited, indent)
    284         previously_visited,
    285         indent,
--> 286         symbols=ascii_symbols
    287     )
    288     return lines

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/display.py in _generate_text_tree(root, show_status, visited, previously_visited, indent, symbols)
    127     if symbols is None:
    128         symbols = unicode_symbols if console.has_unicode() else ascii_symbols
--> 129     tip_id = root.tip().id if root.tip() else None
    130 
    131     def get_behaviour_type(b):

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/composites.py in tip(self)
    102             :class::`~py_trees.behaviour.Behaviour`: the tip function of the current child of this composite or None
    103         """
--> 104         if self.current_child is not None:
    105             return self.current_child.tip()
    106         else:

~/best_data/projects/scratch_python/virtual_env/lib/python3.6/site-packages/py_trees/composites.py in current_child(self)
    669                 if child.status == common.Status.SUCCESS:
    670                     success_children.append(child)
--> 671             return success_children[-1]
    672         else:  # RUNNING
    673             return self.children[-1]

IndexError: list index out of range
```

stonier, 2020-05-11:
*Root Cause*

Parallels (and Sequences, but not Selectors) store an index to the current child while ticking. Then when `tip()` is called, it does an on-the-fly lookup to find the current child via the index. This will fail if the children are added/removed between index generation and lookup.

*Solutions*

Just deal with the ambiguity and return None if there's a problem. That will work around the display problem, however, still leaves a logical ambiguity unresolved at runtime. 

* *Sequence*: `current_child` should update and remain valid if *other* children are added, removed but if *that* child is removed, should it go to None or advance?
* *Parallel*: no change if *other* children are added or removed, but if *that* child is removed, look for another candidate, update the index accordingly

andrewbest-tri, 2020-05-11:
My 2c,

Sequence: Seems fine. Update the index of current child when others are prepended. Though, it won't make any sense because the sequence will go back and run the new higher priority won't it?
Parralel: Does current_child make sense for parallel? If they're all running anyway?

stonier, 2020-05-11:
> sequence will go back and run the new higher priority won't it

You're getting it confused with selectors I think.

> Parralel: Does current_child make sense for parallel? If they're all running anyway?

You're right, it logically should have none (all failing or invalid), one or multiple current children. I made a pragmatic decision there - if there is more than one current child, then it merely points to the last of these. The only use case I know that uses a parallel's current child is `tip()` which is a hint for visualisation or introspection purposes. 

Refactoring the api and visualisation tooling to handle multiple tips is likely to be a complicated affair, so until I know it's needed, I've been putting it off. I really ought to make an issue for that.

stonier, 2020-05-11:
Plurality issue - #288.

closed 2020-05-14 by stonier

stonier, 2020-05-14:
Released in `v2.0.15`.

PR #289 (2020-05-14): Fix current_child problems on dynamic child insertion/removal
* Resolves #287 
* Consolidates code for various types of child removal
* Brings parallel `current_child` calculation into the tick instead of on-the-fly (generates ambiguous results if dynamic insertion/removal happens)