splintered-reality/py_trees #291: Call a subtree with arguments

fedetask, 2020-06-05:
I read through the whole documentation but I didn't really find an answer to this question:
Is it possible to define a sub-tree that takes some parameters as input? One example: a robot may have a `MoveTo(destination)` that can be used many times in the behavior tree, with different locations as `destination` parameter. What is the best way for achieving this?

stonier, 2020-06-06:
Might need you to further explain - your question seems to be asking something different than your examples.

> Is it possible to define a sub-tree that takes some parameters as input?

Generally refer to this pattern as an `idiom`. You can find examples [here](https://github.com/splintered-reality/py_trees/blob/devel/py_trees/idioms.py). When it comes to be robots, we used to define docking as an idiom - it was a subtree composed of actual docking/undocking, led notifications and some simple motion primitives.

> MoveTo(destination) that can be used many times in the behaviour tree

This [example](https://github.com/splintered-reality/py_trees_ros_tutorials/blob/devel/py_trees_ros_tutorials/seven_docking_cancelling_failing.py) has a `FlashLedStrip` instantiated in multiple places in the tree, each with a different colour parameter passed into the constructor.

I wonder if you're looking for something more than these examples though, e.g. dynamic instantiation once you know what `destination` is or reflecting the same behaviour (with it's state) multiple times throughout a tree.

fedetask, 2020-06-06:
Thanks, I did not express myself clearly, but you gave me a good overview. 

> I wonder if you're looking for something more than these examples though, e.g. dynamic instantiation once you know what `destination` is or reflecting the same behaviour (with it's state) multiple times throughout a tree.

Yes, basically the robot will go to different destinations depending on external factors -e.g. if battery low `MoveTo(charging_station)`, if there is an object to pick `MoveTo(object)`, etc. The `MoveTo(destination)` always have the same structure (plans the path, performs the motion commands, etc) but is re-used in different parts of the tree, with a different destination as input.

stonier, 2020-06-07:
Ah, you want to communicate between behaviours. There is the blackboard for that. For this you'll need something like:

* Battery behaviours which catch external battery status and reflect it to the blackboard
* Goal choosing behaviours which read battery status and "write charging_station | object" to the blackboard in a 'destination' variable.
* A `MoveTo` behaviour that reads a `destination` variable from the blackboard. This behaviour would get instantiated twice in the tree, once in the branch that reacts to battery high, the other that reacts to battery low

This is an [example dot graph](https://py-trees-ros-tutorials.readthedocs.io/en/release-2.0.x/tutorials.html#id10) illustrating how blackboard variables help behaviours work together to react to dynamic situations.

This is an [example of a MoveTo behaviour](https://github.com/splintered-reality/py_trees_ros/blob/devel/py_trees_ros/action_clients.py#L34) that picks up it's destination goal from the blackboard. Configured with the appropriate ROS action type, this is essentially a MoveTo style behaviour.

A dot graph would explain that far more easily, but maybe that suffices. Does it help?

fedetask, 2020-06-25:
Yes, it helps! Thank you. Is it recommended to use the blackboard to make something relatively heavy visible from all behaviors, for example a video frame that is updated at every tick?

stonier, 2020-06-26:
It should be fine (sans knowing more details). We've dumped images, even clouds there. There is no difference to sharing a variable out to all your behaviours via some other mechanism or using a variable provided on the blackboard - it's the same consumption in memory. There should be no copy either - the blackboard keys merely point to the objects you pass to it. 
This is all certainly far better than having N behaviours all copying it out of whatever source you have for the video frame within a single tree tick.

Another thing I might add, it's often useful to embed this data gathering activity in a video-frame-to-blackboard behaviour at the front of your tree under a parallel root so it runs on every tick. This has the additional benefit that your tree's decision making is consistent because it's using the same data throughout the tree's tick.

fedetask, 2020-06-27:
Ok, understood. Thank you very much!
I'm closing  the issue since all my questions were answered.

closed 2020-06-27 by fedetask