splintered-reality/py_trees #299: Eternal guard when conditions return Running

fedetask, 2020-07-28:
Hi, 

Is there a way of obtaining the same behavior as EternalGuard, but having conditions that are actually subtrees that can return RUNNING, in which case the right subtree is not ticked, and this "EternalGuard" returns RUNNING?

fedetask, 2020-07-28:
Now that I think about it, what I need is only a Sequence node that does not get stuck in the RUNNING children, but every time it's ticked it executes the sequence from the beginning until one child returns RUNNING or SUCCESS.

How can I obtain such a behavior? Should I create a new Composite?

stonier, 2020-07-28:
You might be able to do that with:

```
+ Parallel (policy=py_trees.common.Parallel.Policy.SuccessOnOne)
  + py_trees.composites.Sequence()
    - Worker 1
    - Worker 2
  - py_trees.behaviours.Success()
```

That will return FAILURE if the sequence returns FAILURE, SUCCESS otherwise. The only caveat is if you're depending on the root of that subtree returning RUNNING, which would be odd, since you're actually terminating execution here.

If that's a common pattern, I'd usually add that to a library as an idiom (see `py_trees.idioms` for examples) rather than a composite, since growing the composite library tends to add complexity to your tree implementations (hard to interpret the logic in a graph if you have N composite types, extra work to make sure visualisation / logging / editor programs are supported).

stonier, 2020-07-28:
What you've described in the second comment above is exactly a selector, so ... questions:

* If the first child fails, does it need to move onto the next child (Selector behaviour) or immediately abort with FAILURE?
* If the first child returns SUCCESS, should it proceed to the next child or return immediately?

stonier, 2020-07-28:
Perhaps:

```
+ Parallel (policy = SuccessOnOne)
  + Selector
    - Child 1
    - Child 2
    - Child 3
  - py_trees.behaviours.Success
```

Here FFF->F, FFR->R, FFS->S, FSI->S, etc.

stonier, 2021-05-10:
With #325 (`v2.1.5`) there are now Sequences w/o memory that can do what you're looking for.

closed 2021-05-10 by stonier