splintered-reality/py_trees #303: sequence, but check nodes every time

jmugan, 2020-10-18:
I'm trying to get the behavior of Figure 1.7 in the book [Behavior Trees in Robotics and AI](https://arxiv.org/pdf/1709.00084.pdf). If something goes wrong in one of the earlier tasks that was initially successful, I want it to go back and do that first, so the sequence starts again at that point. Like, if the robot runs low on battery, I want it to go charge and then go back to the other tasks in the sequence. How do I make Sequence do that? It seems in my testing that once a child returns success, it is not visited again. Is there like a memory setting? I don't want to have to restart the behavior, because this could be a child some larger behavior.

Eternal guard doesn't seem to do what I want either because it stops the behavior instead of just adding a behavior to the top of the stack.

I imagine what I'm asking for is really common, so I am probably missing something simple.

stonier, 2020-10-19:
Wondering if you can provide a concrete description on exactly what you are trying to do, that might help.

> Like, if the robot runs low on battery, I want it to go charge and then go back to the other tasks in the sequence. 

If this is exactly what you're chasing, [idioms.pick_up_where_you_left_off](https://py-trees.readthedocs.io/en/devel/idioms.html?highlight=eternal%20guard#pickup-where-you-left-off) might be the pattern you are looking for.

If you're looking for something that can pop off the last failed behaviour and retreat to one earlier in the sequence (rather than the beginning of the sequence), that might be a bit trickier.

jmugan, 2020-10-19:
Thanks for your response. I'm looking to do an analog of the simpler one, the robot is doing something, get's interrupted with a low battery, and needs to go back to what it was doing before. There is no need to save any complex state of what it was doing before; it just needs to go back to that task. I don't see how the PickupWhereYouLeft will continue until the task is done. If the "High Priority" task Succeeds the whole behavior will Succeed and be done, right? My understanding is that a Selector node Succeeds as soon as one of its children succeeds. I'm just starting with PyTrees, so I am probably misunderstanding something basic.

jmugan, 2020-10-19:
Maybe you could put an inverter decorator on HighPriority task so that if it succeeds, it looks like it failed, and the robot will do the next thing. This means that if it needs to recharge its battery and it is able to do so, the robot can go about its business. But if it fails to recharge its battery, the behavior returns with Success. It seems like this might work, but the semantics seem weird. Hopefully, I'm just misunderstanding something basic.

stonier, 2020-10-19:
You basically want that high priority task to return FAILURE once your battery is high enough and RUNNING | SUCCESS otherwise. 

Usually you'll have it returning RUNNING, with some subtree attached that flashes notificaitons to the users (e.g. led's / sound), moves to dock, idles in the dock. e.g. 

* Selector
  * Parallel (High Priority Subtree Root)
    * Check for battery low (SUCCESS if low, FAILURE if high)
    * Sequence that moves to dock and idles there (idle always returns RUNNING, so this sequence never finishes unless terminated by the check)
  * Actual tasks for `pick_up_where_you_left_of_idiom`

jmugan, 2020-10-19:
Thanks! Let me think about that for a bit.

closed 2020-10-19 by jmugan

stonier, 2020-10-24:
Shouldn't be too confusing here - just a 'guard' that returns failure if the battery gets high enough and a subtree that is RUNNING while it's recharging. I usually name that Parallel something like "Need Recharging?" and then it's easier to grokk.

You will find the logic sometimes confusing getting started though. I always found myself having to invert my natural boolean intuition to trigger the correct pathways in the tree but after a while, you stop thinking of SUCCESS & FAILURE and more of GREEN LIGHT/RED LIGHT (GO/STOP). At the end of the day, planning out a tree is an exercise in defining pathways and GREEN LIGHT/RED LIGHT feels like a more natural fit for the problem.

jmugan, 2020-10-24:
Makes sense. Thanks for your help!