splintered-reality/py_trees #313: Sequences w/o Memory, Selectors w/ Memory

stonier, 2021-02-08:
Thread for designing and tracking an implementation of these in `py_trees.` NB: the current composities include Sequences w/ Memory and Selectors w/o Memory.

There have been numerous requests for these and at this point, the surprise at not supporting these paradigms is probably greater than the additional complexity it introduces. That complexity shouldn't be too great if handled in an optimal way.

For background reading, see #85, #202, #311.

I'll include some design notes here before I implement. Please watch/comment in the thread here if you have requirements beyond those mentioned already.

----

wafts95, 2021-02-08:
Hi @stonier, do you have some preferred/ defined contributing guidelines? I may have some work done in the direction of this Issue that we could use.

stonier, 2021-02-11:
> Hi @stonier, do you have some preferred/ defined contributing guidelines? I may have some work done in the direction of this Issue that we could use.

On request - PR https://github.com/splintered-reality/py_trees/pull/315!

Let me know if you have questions.

sachit-tri, 2021-02-12:
I support this noble endeavor. Looking forward to the design notes.

stonier, 2021-02-13:
Design question: should the selector advance or stop if a running child fails? e.g.

```
# TICK 1
Selector w/ Memory (R)
 - Foo (R)
 - Bar (-)
 
 # TICK 2a - Advance
 Selector w/ Memory (R)
  - Foo (F)
  - Bar (R)
  
  # OR TICK 2b - Stop
  Selector w/ Memory (F)
   - Foo (F)
   - Bar (-)
```

The original Chooser in this library stopped. @jbcpollak advances with his implementation in PR #312. Does anyone here have a use case that doesn't advance? If you're familiar with other tree libraries, what do they typically do with their implementations?

jbcpollak, 2021-02-13:
I don't understood the 2b option. How would that work with multiple options? When does ever advance to Bar?

We did have a use case were we wanted to start down one path, but then if it failed _not_ try the alternative path. I don't think this is equivalent to the 2b option either - we solved this with a sequence and inverted guards on each child. That feels more like an idiom than a new type of composite though

stonier, 2021-02-13:
2b is exactly as you described. If, in the above example, `Foo` had failed immediately, it would have advanced to `Bar`.

I feel like 2a is suprising. If there had been higher priorities preceding `Foo`, they would not have been explored due to the memory, but it permits exploration of those lower priorities.

I feel like 2b is also surprising though. A selector which doesn't keep falling back to lower priorities.

jbcpollak, 2021-02-13:
I have always referred to this article for how BTs should work, although there's obviously a lot of variation and different needs

https://www.gamasutra.com/blogs/ChrisSimpson/20140717/221339/Behavior_trees_for_AI_How_they_work.php

The article also references a Java implementation that I have looked it - boggles my mind why not.

andrewbest-tri, 2021-02-15:
I don't really grok the use-case of the selector with memory, though I was adamant about sequences without memory :-). I would expect 2a to be the default though. It keeps the concept of "memory" a little more consistent. In other composites, as I recall, memory means the composite remembers the last child it ticked but didn't change the fundamental behavior of the composite aside from that. 2b changes what the selector does.

sachit-tri, 2021-02-15:
BTW I'm not understanding how state 1 is even valid?

> \# TICK 1
> Selector w/ Memory (R)
> \- One (S)
> \- Foo (R)
> \- Bar (-)

Shouldn't the Selector return Success as soon as One returns SUCCESS. Did you mean you want to use something like this?

# TICK 1
Selector w/ Memory (R)
 - One (F)
 - Foo (R)
 - Bar (-)

And on Tick 2, start ticking at Foo instead of One. If Foo returns Failure, move on to tick Bar. If Foo returns Success, then return Success.

jbcpollak, 2021-02-16:
@sachit-tri - yes, you are right, that is the (well, _my_ I guess, lol) expect behavior. You are right the OP has a typo, if One returns (S) the selector is done.

@andrewbest-tri, re: 

> I don't really grok the use-case of the selector with memory, though I was adamant about sequences without memory :-)

Our usecase is something like:

Selector w/ Memory (Maybe "ResumingSelector" or something is a better name? IDK)
* FindTheThing
* GetSatelliteReconnaissance
* RunAndHide

So basically, try and find the thing. You'll stay in `FindTheThing (R)` for multiple ticks. Eventually the subtree gives up and returns FAILURE, so you `GetSatelliteReconnaissance(R)`, which if it succeeds is maybe an alternate way to solve the overall problem, and lastly if that fails, you run away (which in some cases we force to always fail so if you got that far the Selector is known to fail.

We are doing robotics, not Game AI, so this is an analogy, not directly relevant.

stonier, 2021-02-22:
> @sachit-tri - yes, you are right, that is the (well, my I guess, lol) expect behavior. You are right the OP has a typo, if One returns (S) the selector is done.

Aye, oops on my part. Corrected to (F).

wafts95, 2021-03-28:
Hi, sorry for being late to the party. 

> Design question: should the selector with memory advance or stop if a running child fails? e.g.

The classical formulation of BTs theory clearly defines the behavior of Selector and Sequence nodes, whether or not they have memory. In (1) this classical formulation is explained, which states that a Sequence node Succeeds if all of its children Succeed, and Fails if any one of its children Fail. A Selector node Succeeds if any one of its children Succeed, and Fails if all of its children Fail. There is no mention of the Running status of the current child, because all what matters is either the Success or Failure of the child. See Table below.

[image]

Therefore, whether or not a certain child of a Selector fails upon startup (immediately returns Fail), or during execution (after retuning Running), it shouldn't matter as to how the children are exactly executed; we should simply follow the basic definition of "all children" or "one child" Succeed/Fail, and this means going for option 2a.

And to further illustrate why we ought to not track if the child returns Fail instantly, consider the analogy to the Sequence node's operation: when the child of a Sequence node Succeeds, to decide to move to the next child, does it matter if the the current child had Succeeded instantly, or had Succeeded after Running for a while? The obvious answer is no (at least to me). We always move to execute the subsequent child anyway.

So, I recommend we go with 2a to follow the Classical BT formulation and avoid having a special formulation, and to make sure that this library can be adopted by the largest possible audience.

Reference:
[1]	M. Colledanchise and P. Ögren, Behavior Trees in Robotics and AI: An Introduction. Chapman & Hall/CRC Press, 2018.

jbcpollak, 2021-03-28:
> Hi, sorry for being late to the party.

@wafts95 this is my view as well, thank you for explaining it more clearly than I could.

stonier, 2021-05-04:
On 2a) (when a running child of a selector fails, proceed to investigate the lower priorities) - agreed.

Though not for quite the same reason. I'm not sure that holding to a classical definition when a new use case arrives is the quite the right justification. Not listening to the actual problems-to-solve can lead to irrelevance. For now though, the folks I know with the use cases for 2b) have moved on, so that leaves 2a) to be served. That's good enough for me. The fact that it still falls neatly under the classical definitions is a bonus.

At any rate, PR is up for the selector-side of this if you care to look (#324). 

Apologies for the delay - family and winter (:snowboarder:) pulled me away for a while!

stonier, 2021-05-10:
With #324 and #325, this has slipped in as a patch release in the `v2.1.x` branch (`v2.1.5`) using defaults that don't break api for current users. I'll let this iterate for a while before releasing `v2.2.x` with better defaults, #326.

@jbcpollak and @wafts95, thanks for your input to help steer this!

closed 2021-05-10 by stonier