splintered-reality/py_trees #323: Visualization Tool like Groot

NickNair, 2021-04-27:
Is there any visualization tool that we can directly integrate with our py_trees implementation. I'm looking for something like groot that is integrated with the Behavior-Tree.CPP package.

sachitv, 2021-04-27:
You can use https://github.com/splintered-reality/py_trees_js

stonier, 2021-04-27:
If you're already in a ROS environment, you can make use of the ros implementation of `py_trees_js` in https://github.com/splintered-reality/py_trees_ros_viewer.

OR use that as a template to roll `py_trees_js` into an application (qt, web, ... ) of your own.

NickNair, 2021-04-27:
Isn't that package for ROS2. I was looking for something compatible with  ROS noetic.

stonier, 2021-04-27:
A little bit clunkier and doesn't give you the blackboard but https://github.com/splintered-reality/rqt_py_trees gives you the equivalent runtime viewer in ROS1.

closed 2021-05-10 by stonier