splintered-reality/py_trees #334: create_behaviour_from_function spec'd to return `Any`

jbcpollak, 2021-06-01:
calling `create_behaviour_from_function` like this:

```
        def update(self: py_trees.behaviour.Behaviour) -> py_trees.common.Status:
                return py_trees.common.Status.SUCCESS

        return py_trees.meta.create_behaviour_from_function(
            update
        )(name='Has More To Do')
```

results in this error from my_py:
` error: Returning Any from function declared to return "Behaviour"`

Looking at how `create_behaviour_from_function` uses the `type` function, I suspect this is more a mypy issue than py-trees, but it might be nice to explicitly declare the fuction return a `Behavior`

jbcpollak, 2021-06-02:
came across another example that causes the same error:

`return py_trees.behaviours.Failure()`

results in:

`error: Returning Any from function declared to return "Behaviour"`

stonier, 2023-01-18:
```
def create_behaviour_from_function(
    func: typing.Callable[[typing.Any,], common.Status],
    module: typing.Optional[str] = None
):
) -> "typing.Type[behaviour.Behaviour]"
```

in #380.

closed 2023-01-18 by stonier