splintered-reality/py_trees #335: class Composite should explicitly declare current_child type

jbcpollak, 2021-06-01:
class Composite should be declared as:

```
class Composite(behaviour.Behaviour):
    current_child: Optional[Behaviour]
```

otherwise current_child is typed as None according to the Composite.__init__ method.

jbcpollak, 2021-06-02:
update: I discovered this happens with the mypy rule `check_untyped_defs = True` enabled in a project. I'm going to look more into that rule and if it can be tweaked on a per-dependency basis.

jbcpollak, 2021-06-02:
looking into the `check_untyped_defs` option, I definitely think its ideal to enable, and it does warn about this problem, as well as others. I can make a PR to clean up the errors it causes.

https://mypy.readthedocs.io/en/stable/config_file.html#confval-check_untyped_defs

stonier, 2023-01-22:
Thanks for the heads up!

Fixed in #380.

closed 2023-01-22 by stonier