splintered-reality/py_trees #336: ComparisonExpression could probably use Generics

jbcpollak, 2021-06-01:
I think ComparisonExpression could benefit from generics to improve type specificity. What happens now is that because operator is declared to take two `Any`s, mypy returns:

` error: Returning Any from function declared to return "Behaviour"`

for a lot of common use cases, because the variable and value are both `Any`.

I think defining the class like so might help (I can try and test in the future, just wanted to flag it now):

```
T_contra = TypeVar('T_contra', contravariant=True)  # contravariant.
V_contra = TypeVar('V_contra', contravariant=True)  # contravariant.

class ComparisonExpression(Generic[T_contra, V_contra]):
    """
    Store the parameters for a univariate comparison operation
    (i.e. between a variable and a value).

    Args:
        variable: name of the variable to compare
        value: value to compare against
        operator: a callable comparison operator

    .. tip::
        The python `operator module`_ includes many useful comparison operations, e.g. operator.ne
    """
    def __init__(
        self,
        variable: str,
        value: V,
        operator: typing.Callable[[T_contra, V_contra], bool]
    ):
        self.variable = variable
        self.value = value
        self.operator = operator
```

I haven't tested the above, but it should work. The contravariant part may not be strictly necessary but if users have cases where "variable" or value is sometimes a child type of the declared ComparisonExpression, it would be helpful.

jbcpollak, 2021-06-01:
As an aside, what's the value of having the `value` variable at all? Since its fixed, why not just have the operator take a single parameter, the current value of `variable`? IE, what's the difference between:

```
            behaviours.CheckBlackboardVariableValue(
                check=ComparisonExpression(
                    variable='check_var',
                    value=20,
                    operator=lambda variable, value:
                        variable != value,
                )
             )
```

and:

```
            behaviours.CheckBlackboardVariableValue(
                check=ComparisonExpression(
                    variable='check_var',
                    value=20,
                    operator=lambda variable, value:
                        variable != 20,
                )
             )
```

also BTW, if `value=None`, then we'd want to us `is not None` to be correct, so I've started doing things like:

```
            behaviours.CheckBlackboardVariableValue(
                check=ComparisonExpression(
                    variable='check_var',
                    value=None,
                    operator=lambda variable, value:
                        variable is not None,
                )
             )
```

which makes mypy happy but seems redundant.

stonier, 2023-01-22:
Closing out for now (see notes in #386). Expecting this may come resurface though, please re-open if necessary.

Thanks for raising the problem!

closed 2023-01-22 by stonier