splintered-reality/py_trees #340: Sequence class (>2.1.4) adds characters which are not accepted by codecs

rudyg-123, 2021-06-08:
[image]

When using the render_dot_tree function under py_trees.display, I encountered an issue whereby the writer could not successfully parse Sequence classes. I created a graph with a simple Behaviour Tree for rendering and it continuously failed unless reverted back to version 2.1.4 of py_trees. See the error below.

rudyg-123, 2021-06-08:
[image]

stonier, 2021-06-08:
Ah, yes. I added some additional symbols to distinguish between variants of parallels, sequences and selectors in c24ceace, but forgot to call `has_unicode()` first to make sure it would gracefully fallback to ascii symbols like the text tree displays do.

Definitely a bug. Giving it priority.

stonier, 2021-06-08:
Q: Does your console fail on all unicode symbols or does it only fail with circled-M?

You can check by navigating to the `py_trees` package, wherever it is on your system (don't need to care about paths) and execute:

```
cd <path-to>/py_trees
python3 ./console.py
```

rudyg-123, 2021-06-09:
[image]

stonier, 2022-06-07:
To reproduce:

Setup a non utf-8 locale
```
$ cat /usr/share/i18n/SUPPORTED | grep en_US
$ sudo locale-gen en_US.ISO-8859-15
$ sudo update-locale
$  locale -a
```

Confirm that unicode is not available:
```
$ python3 ./console.py
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
dude
[DEBUG] loginfo message
[ WARN] logwarn message
[ERROR] logerror message
[FATAL] logfatal message
red
some normal text
    Name: Dude
special characters are

Traceback (most recent call last):
  File "./console.py", line 343, in <module>
    print("lightning_bolt: {}".format(lightning_bolt))
  File "/usr/lib/python3.8/encodings/iso8859_15.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u26a1' in position 16: character maps to <undefined>
```

Then call the render demo to tickle the error.
```
py-trees-demo-dot-graphs 

*******************************************************************************
                                   Dot Graphs                                  
*******************************************************************************

Renders a dot graph for a simple tree, with blackboxes.

    Generate Full Dot Graph

        py-trees-demo-dot-graphs

    With Varying Visibility Levels

        py-trees-demo-dot-graphs --level=all
        py-trees-demo-dot-graphs --level=detail
        py-trees-demo-dot-graphs --level=component
        py-trees-demo-dot-graphs --level=big_picture

*******************************************************************************

Writing /mnt/mervin/workspaces/devel/py_trees/src/py_trees/py_trees/demo_dot_graphs_fine_detail.dot
Traceback (most recent call last):
  File "/mnt/mervin/workspaces/devel/py_trees/src/py_trees/.venv/bin/py-trees-demo-dot-graphs", line 11, in <module>
    load_entry_point('py-trees', 'console_scripts', 'py-trees-demo-dot-graphs')()
  File "/mnt/mervin/workspaces/devel/py_trees/src/py_trees/py_trees/demos/dot_graphs.py", line 121, in main
    py_trees.display.render_dot_tree(root, args.enum_level)
  File "/mnt/mervin/workspaces/devel/py_trees/src/py_trees/py_trees/display.py", line 687, in render_dot_tree
    writer(pathname)
  File "/mnt/mervin/workspaces/devel/py_trees/src/py_trees/.venv/lib/python3.8/site-packages/pydot.py", line 1826, in write
    f.write(s)
  File "/usr/lib/python3.8/encodings/iso8859_15.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u24c2' in position 344: character maps to <undefined>
```

PR #359 (2022-06-07): [display] prevent unicode leakage when no unicode for dot rendering
Resolves #340.
commit 8d8fc19d (2022-06-07): [display] prevent unicode leakage when no unicode for dot rendering

Resolves #340.
closed 2022-06-07 by stonier