splintered-reality/py_trees #341: What version to use: Python 3.x, Ubuntu 20.04

pitosalas, 2021-07-11:
Which of the versions should I use? I will be using it with ROS1 Noetic but I believe your ROS support is all on ROS2. That's fine. But for Ubuntu 20.04 and Python 3.x, should I use py-trees 1.0 or some other one? Thanks!

stonier, 2021-07-12:
It all depends on what your use case on 20.04/python3 is. For ROS related requirements, do look up the [py_trees_ros/README](https://github.com/splintered-reality/py_trees_ros/blob/devel/README.md).

* ROS 1 Noetic &rarr; `py_trees 0.7`, `py_trees_ros 0.6`, `rqt_py_trees 0.4`, ...
* ROS 2 Foxy &rarr; `py_trees 2.1`, `py_trees_ros 2.1`, `py_trees_ros_viewer 0.2`, ...
* No ROS &rarr; `py_trees >0.7` will do. You might choose `0.7` for compatibility for instance before making a large jump to `2.1`.

pitosalas, 2021-07-12:
> 	• ROS 1 Noetic → py_trees 0.7, py_trees_ros 0.6, rqt_py_trees 0.4, ...
> 	• ROS 2 Foxy → py_trees 2.1, py_trees_ros 2.1, py_trees_ros_viewer 0.2, ...
> 	• No ROS → py_trees >0.7 will do. You might choose 0.7 for compatibility for instance before making a large jump to 2.1.

Hi Daniel

Thanks, very useful. BTW what do you mean "for compatibility" ? With what?

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2021-07-12:
Hi Again

When I install I am unable to install on Python3.7, Noetic. Any quick tip?

and

Pito Salas
Faculty, Computer Science
Brandeis University

stonier, 2021-07-12:
> compatibility

If you had existing code running on `0.6` with python 2, you might not want to make the big jump to `2.1` immediately.

> unable to install on Python3.7, Noetic. Any quick tip?

You might be missing some information in your post above - I'm getting blanks before, after the `and`. How did you try and install? What errors are you seeing?

pitosalas, 2021-07-12:
Thanks!!

Oh. I had a couple of images in there. Let me type in the text. Sorry.

Pip install py_trees_ros -> a long stack trace ending with client error: not found for url: ...

And 

Pip install rqt_py_trees -> Requirement already satisfied in rqt_py_trees in /... melodic ... even though I have the default python set to 3.x and noetic as the default. 

Now, if my goal is indeed to work with Ros, do you *strongly* recommend my studying py_trees_ros or is it more like some syntactic sugar? I.e. will I learn important temhcnicales and patterns? (I don't plan to go to ROS2 any time soon.!)

Thanks!

Pito Salas
Faculty, Computer Science
Brandeis University

stonier, 2021-07-13:
**Pip**

* If you're just doing python-only work, `py_trees` from pip is a good solution (none of the ros packages are on pip)
* If you're in a ROS workspace, don't use pip, fetch them all from ROS debs

**py_trees_ros e.t. a.l.**

There are two reasons you might want to use these.

**Ros Behaviours** - to be honest, there's not much here and you could recreate your own behaviours using these as inspiration pretty easily. Most noteworthy - generic behaviours for handling subscribers and action clients in useful ways.

**Debugging and Visualisation Tools**

These are super-useful even if the ROS1 versions of these tools are rather clunky. Easy to use to echo the contents of the blackboard or the tree state. You can also bag the tree state and replay it in the viewer. 

Caveat - there was an update in ROS1 not so long ago that thwarted use of ascii color codes in ros topics. I'm not sure if I updated the ROS1 code to handle that. If it's spitting our ascii color code sequences, let me know. That's maybe something worth fixing.

stonier, 2021-07-13:
If you skim through the [noetic tutorials](http://docs.ros.org/en/noetic/api/py_trees_ros/html/tutorials.html) you'll get a better idea of what's worth chasing.

pitosalas, 2021-07-14:
The link is to a page that talks about kinetic. I assume that's still the right page, yes?

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2021-07-14:
I'm really sorry about these dumb questions but I am trying to install it correctly.

Now on a working Noetic installation I get Unable to locate package for:

sudo apt install ros-noetic-py-trees
sudo apt install ros-noetic-py-trees-ros
sudo apt install ors-noetic-rqt-py-trees

And all of those, chasing the last two dashes to underscores. I've done an sudo apt-get update.

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2021-07-15:
I figured it out. Sorry for the bad question.

closed 2021-07-15 by pitosalas