splintered-reality/py_trees #344: Changes to action.py

tejalbarnwal, 2021-12-02:
If I try to add/ modify the `py_trees/demos/action.py` it doesn't get reflected in the terminal output? Do I have to run the make file? How can I reflect on those changes?
Further, cant we run the .py file here by just doing `python action.py`? If not, why so?
I am sorry for asking such dump questions, but I am just a beginner.

stonier, 2021-12-03:
> If I try to add/ modify the py_trees/demos/action.py it doesn't get reflected in the terminal output? Do I have to run the make file? How can I reflect on those changes?

Depends how you're consuming that module. Are you running `. ./venv.bash` and then `py-trees-demo-action-behaviour`? That should catch your changes.

If you've some other workflow, python cache files might be getting in the way, or you might have another py-trees installed on your system that is getting in the way.

> Further, cant we run the .py file here by just doing python action.py? If not, why so?
I am sorry for asking such dumb questions, but I am just a beginner.

It's got no entry point to be an executable script. By that I mean, you need something like

```
if __name__ == "__main__":
    main()
```

to turn it into an executable script. Not needed here, the setuptools creates these for you - [here](https://github.com/splintered-reality/py_trees/blob/devel/setup.py#L60).

tejalbarnwal, 2021-12-03:
@stonier that solves my problem. Thank you so much!

closed 2021-12-03 by tejalbarnwal