splintered-reality/py_trees #417: Blackboard watcher ERROR: service not found

RobertoHernandezjr, 2023-03-17:
When trying to launch the blackboard tutorial here:
https://py-trees-ros-tutorials.readthedocs.io/en/release-2.1.x/tutorials.html

If you try:
py-trees-blackboard-watcher

You get :
ERROR: service not found [type: py_trees_ros_interfaces/srv/GetBlackboardVariables]

However if you try any of the other commands with option:
py-trees-blackboard-watcher /battery.percentage

It works, and if you go back and try to watch all variables by using 'py-trees-blackboard-watcher'
after now you dont get any errors and it works.

stonier, 2023-03-26:
So I can reproduce:

Which tutorial were you failing on?
Which ROS release? Or was it from sources?

RobertoHernandezjr, 2023-03-26:
Tutorial 3 - Introspect the Blackboard
Ros Foxy
Unbuntu 20.04