splintered-reality/py_trees #420: Blocking behaviours question

DarkJoker9817, 2023-04-19:
Hello, thanks for your work on this library, I have a doubt about the execution of the tree. What if a custom Behaviour is blocking, for example when in the update method there is a call to a ROS service that takes longer to execute than the tree tick period? What is the 'behaviour' of the tree in this case?

DarkJoker9817, 2023-04-21:
OK thank you very much, I'm already using action clients and pub-sub, mine was just a curiosity about how the behavior of the tree might evolve in that case. In which class I can find tree manager?

stonier, 2023-09-05:
Very late reply sorry, but you'd be looking for the ROS tree manager, look at [py_trees_ros.trees.BehaviourTree](https://github.com/splintered-reality/py_trees_ros/blob/2.2.2/py_trees_ros/trees.py).

closed 2023-09-05 by stonier