splintered-reality/py_trees #441: MInor code mistake in documentation

Jibrangit, 2024-03-21:
I am not sure how this happened, but on the demos page http://docs.ros.org/en/melodic/api/py_trees/html/demos.html#id1 , in the code used for the various demos, the lines of code for returning the status is invalid. Instead of `new_status = py_trees.Status.SUCCESS` it should be new_status = `py_trees.common.Status.SUCCESS`. 

This is ofcourse the case in the script that's located in the pip package and the script that executes the main() in action.py runs as expected. I only discovered this error because I copy-pasted the code and ran it as I wanted to tweak around.

Jibrangit, 2024-03-21:
Actually, proceeding up the demos, I find plenty of mistakes in the code used in the documentation. At this point, I would put a disclaimer to NOT COPY the code directly from the documentation webpage.

sea-bass, 2024-08-31:
Rather than a disclaimer, you're welcome to contribute fixes to the documentation as you see them! That would be much more helpful.

naveedhd, 2024-12-30:
@Jibrangit the demo: http://docs.ros.org/en/melodic/api/py_trees/html/demos.html#id1 is for py_trees version 0.6.9 and for that version the demo code works fine as in the readme. i.e. `py_trees.Status.SUCCESS` is valid because it is [exported](https://github.com/splintered-reality/py_trees/blob/release/0.6.x/py_trees/behaviour.py#L25).

closed 2026-09-10 by sea-bass